rust-embedded/book #4: Book: Section 2: Blinking your first LED (draft)

japaric, 2018-08-10:
_From @jamesmunns on July 10, 2018 9:32_

This section covers the equivalent of getting "Hello World" working for an embedded target. Examples should start high level, and build back up towards that point.

_Copied from original issue: rust-embedded/wg#117_

japaric, 2018-08-21:
Triage: [Tthis section](https://github.com/rust-embedded/book/issues/4) has no instructions. @jamesmunns could you add some? IIRC, we agreed on using `cortex-m-rt` as a base and covering how to write a program in two different ways: using a high level api (e.g. `f3`) crate and w/o external dependencies.

japaric, 2018-08-27:
I think it would be good to start this section with instructions on how to build and debug a Cortex-M program running on QEMU since everyone should be able to run that without external hardware. Then we can move onto the discovery board.

WojciechMigda, 2018-08-28:
@japaric , would this be of any help?
https://github.com/WojciechMigda/how-to-qemu-arm-gdb-gtest

japaric, 2018-08-29:
@WojciechMigda Unfortunately, no. That's using QEMU userspace emulation (`qemu-arm`) and that can only run programs compiled for hosted environments like the `arm-unknown-linux-gnueabi` target; that can't execute (or won't properly execute) bare metal programs compiled for ARM Cortex-M.

What we have to use in this section is QEMU full system emulation (`qemu-system-arm`). That can emulate a Cortex-M3 core (the LM3S6965) so we can run programs compiled for the `thumbv7m-none-eabi` target on it. I have done some initial experiments with that in [this repo](https://github.com/japaric/lm3s6965evb) and have recently confirmed that semihosting works so we can cover hello world (see `examples/hello.rs` in the cortex-m-quickstart repo) in this chapter.

frafra, 2018-08-29:
Just wanted to thank you for your great work. I am still learning Rust, I ordered the board you suggest, and I am eager to follow your tutorials. I hope I will be able to give something back to the community in the future. Thank you! :smiley:

japaric, 2018-11-20:
This is pretty much done.

closed 2018-11-20 by japaric