rust-embedded/book #5: Book: Section 3A: Static Guarantees (draft)

japaric, 2018-08-10:
_From @jamesmunns on July 10, 2018 9:33_

What sort of Static Guarantees are possible with Rust?

_Copied from original issue: rust-embedded/wg#118_

japaric, 2018-08-10:
Static guarantees is a topic that's too broad. This section was originally brought up in the context of "the static configuration pattern with embedded-hal implementation as an example".

So the section should probably be named "static configuration" or "type state applied to configuration" or just "type state applied to embedded".

jamesmunns, 2018-08-21:
I have all of my notes from my RustConf talk that covered this. I plan to at least dump them all here today.

In particular, my talk addressed use of the type system, including zero sized typed, to enforce design contracts, preconditions, and postconditions.

japaric, 2018-12-11:
The part about type state configuration has been covered.

closed 2018-12-11 by japaric