rust-embedded/book #41: Use Rust 2018 syntax?

thejpster, 2018-10-10:
Should we be using Rust 2018 syntax in the book? The main change would be removing `extern crate` statements.

hyperslv, 2020-06-12:
agree but the suggested change has to be synchronized with the appropriate change in source code of the examples at github.com/rust-embedded/cortex-m-quickstart .
I will help & fix both if someone explains me how to synchronize these Repos

therealprof, 2020-06-12:
What do you mean by synchronize? Open PRs on both and then we'll merge them, couldn't be easier. 😅

hyperslv, 2020-06-13:
Created a [PR](https://github.com/rust-embedded/book/pull/248) on this issue

hyperslv, 2020-06-15:
This issue might be closed since the appropriate [PR](https://github.com/rust-embedded/book/pull/248) was merged.

Narukara, 2023-01-17:
There are still some `extern crate` statements in the book. For example:

https://github.com/rust-embedded/book/blob/19f798d448835a4888e3b3eae7fe69f1d61d8681/src/peripherals/singletons.md?plain=1#L65

https://github.com/rust-embedded/book/blob/09986cd352404eb4659db44613b27cac9aa652fc/src/collections/index.md?plain=1#L33

Shall we fix them all?

eldruin, 2023-01-17:
@Narukara I would welcome a PR fixing them all. We could then close this issue.

commit c7e79031 (2023-01-18): Merge #340

340: remove extern crate statements r=eldruin a=Narukara

from issue #41 

I kept this one because `alloc` can only be imported that way:

https://github.com/rust-embedded/book/blob/19f798d448835a4888e3b3eae7fe69f1d61d8681/src/collections/index.md?plain=1#L33

Closes #41
PR #340 (2023-01-18): remove extern crate statements
from issue #41 

I kept this one because `alloc` can only be imported that way:

https://github.com/rust-embedded/book/blob/19f798d448835a4888e3b3eae7fe69f1d61d8681/src/collections/index.md?plain=1#L33

Closes #41
closed 2023-01-18 by bors[bot]