rust-embedded/book #46: Editing: Merge sections on Peripherals?

jamesmunns, 2018-10-15:
Currently we discuss peripherals/memory mapped registers in two places:

1. https://github.com/rust-embedded/book/blob/master/src/start/registers.md
2. https://github.com/rust-embedded/book/tree/master/src/peripherals

I think there is a little bit of overlap here, but generally the second portion covers a little more theory on why we don't do it like C does, but I haven't written the part that explains what we actually do instead. Instead there is just [a placeholder](https://github.com/rust-embedded/book/blob/master/src/peripherals/rusty.md). I think portion 1 above is generally most of what I wanted to cover in this section.

CC @thejpster for thoughts

thejpster, 2018-10-16:
lol. What's happened here is I've totally misunderstood the brief of "semihosting and register access".

thejpster, 2018-10-16:
Let's move start/registers.md to src/peripherals. @japaric can you clarify what you'd like me to cover in the "start" section?

japaric, 2018-11-06:
Sorry for taking so long to look at this. I agree that we should move start/registers.md to src/peripherals. I think it should go first so the peripherals chapter should cover: mmio (volatile op), structs (repr(C)) and then "the case for singletons", which is what src/peripherals currently has.

The getting started chapter is about "how do I" and "what can I do with this". Later chapters cover "how does this works" and "why does it look like this". So start/peripherals.md should cover one of the cortex-m peripheral APIs, e.g. peripheral::SYST. Specifically it should cover: taking peripherals into the current scope (skipping explaining why is done like that), the low level API (read / write on individual registers: e.g. SYST.rvr.write) and the high level API (SYST.enable_counter, which is just some registers reads / writes).

thejpster, 2018-11-09:
@japaric, is https://github.com/rust-embedded/book/pull/76 heading in the right direction?

japaric, 2018-11-20:
This was done in #76.

closed 2018-11-20 by japaric