rust-embedded/book #48: I have a question, oh also, where do I ask questions?

freiguy1, 2018-10-16:
When it comes to embedded development, I'm fairly used to not debugging. On a related note I'm _not_ used to gdb. Is there any way I can build with cargo and then just use openocd to push the compiled bytes to my device? All the documentation I've seen requires gdb to flash the program to the board.  Would this be worth adding to the book?

Secondly, where do I ask questions about embedded rust dev? Making an issue like this seems a bit clunky, but I couldn't find any info related to this topic either.

adamgreig, 2018-10-16:
Hi!

To answer your second question first: good places include the [embedded category on the forum](https://users.rust-lang.org/c/embedded) and the IRC (#rust-embedded @ irc.mozilla.org).

I'd really recommend trying out debugging on embedded, especially with gdb. It can be a really quick and powerful way to solve problems. Even if you're not going to use it for debugging though, you do have to use it to load code via OpenOCD. It should be as simple as `target :3333` followed by `load`, but the various tutorials and readmes are probably a better starting place.

If you just want to push a compiled binary to your device, you can actually skip OpenOCD entirely and use vendor supplied tools such as STLink (or the Linux-compatible [texane/stlink](https://github.com/texane/stlink)).

There is work ongoing on a [new book](https://github.com/rust-embedded/new-book/issues/1) which will cover programming and debugging in detail, with lots of specific examples for different pieces of hardware.

freiguy1, 2018-10-16:
Thanks for the quick and helpful response. I think my main problem is inexperience with gdb. I was able to get a blink program with RTFM flashed to my blue pill with gdb. It just seemed cumbersome to build, run gdb, then type `continue`, `next`, `next` to get my program running each time I make a change. Also it seems I'm not exiting gdb correctly because my blinking led stops blinking. I'm being intentionally vague here because I can't remember each detail since it's been a few weeks.

adamgreig, 2018-10-16:
Are you using semihosting (or panic-semihosting)? Semihosting will generally require an attached debugger to function, and otherwise blocks or panics.

freiguy1, 2018-10-18:
@adamgreig aww - I didn't see your reply. I absolutely am and that was the issue. I just figured it out and was hoping to enlighten everyone :)

Thanks again!

closed 2018-10-18 by freiguy1