rust-embedded/book #52: Chapter Title Freeze for Rust Website

jamesmunns, 2018-10-16:
On the upcoming Rust website revision, we link to sections of the embedded book. We need to finalize/freeze these chapter names, so the links will be consistent (for now):

* Static Guarantees
    * https://github.com/rust-embedded/book/issues/5
    * https://rust-embedded.github.io/book/static-guarantees/static-guarantees.html
* Dynamic Data Structures/Flexible Memory Management
    * https://github.com/rust-embedded/book/issues/8
    * https://rust-embedded.github.io/book/collections/collections.html
* Concurrency
    * https://github.com/rust-embedded/book/issues/7
    * https://rust-embedded.github.io/book/concurrency/concurrency.html
* Interoperability
    * https://github.com/rust-embedded/book/issues/1
    * https://rust-embedded.github.io/book/interoperability/interoperability.html
* Portability
    * https://github.com/rust-embedded/book/issues/6
    * https://rust-embedded.github.io/book/portability/portability.html

jamesmunns, 2018-10-16:
CC @rust-embedded/resources

jamesmunns, 2018-10-16:
I should note that the most important things to stabilize here are:

* the filenames of the markdown file related to these chapters
* any folder names above these markdown files that are reflected in the URL

thejpster, 2018-10-19:
Could I propose we don't put `.html` on the end of the canonical URLs? URLs are forever, file formats are for Christmas (or something).

thejpster, 2018-10-19:
Also, for consistency, maybe we should rename #8 to 'Concurrency'? (Or rename the directory to dynamic_data_structures?)

adamgreig, 2018-10-21:
> Also, for consistency, maybe we should rename #8 to 'Concurrency'? (Or rename the directory to dynamic_data_structures?)

'Collections'?

jamesmunns, 2018-10-30:
@thejpster I do not believe it is possible to omit the `.html`. 

For example:

* `https://rust-embedded.github.io/book/intro/no-std.html`
    * [works](https://rust-embedded.github.io/book/intro/no-std.html)
* `https://rust-embedded.github.io/book/intro/no-std`
    * [doesn't work](https://rust-embedded.github.io/book/intro/no-std)
* `https://rust-embedded.github.io/book/intro/`
    * [doesn't work](https://rust-embedded.github.io/book/intro/)

jamesmunns, 2018-10-30:
@rust-embedded/resources 

So I would propose the following:

1. Rename #8 to `Collections`, to match the current book contents and link
2. use links in the format `https://rust-embedded.github.io/book/intro/no-std` for the website

This leaves one open question IMO:

Do we want to use the `book` link, or the `bookshelf` link? Concretely:

* `https://rust-embedded.github.io/bookshelf/book/intro/no-std` OR
* `https://rust-embedded.github.io/book/intro/no-std`

If we decide on this today, I will fix the links in the website, and I will add comments in the `SUMMARY.md` mentioning that the layout and links must not change without fixing upstream first (or having legacy redirects).

thejpster, 2018-10-30:
I vote https://rust-embedded.github.io/book/intro/no-std

japaric, 2018-10-30:
> Do we want to use the book link, or the bookshelf link? 

At the end the rust-embedded.org link will be https://docs.rust-embedded.org/book/intro/no-std, right? So there won't be a bookshelf (or docs) in the path.

jamesmunns, 2018-11-13:
This can probably be closed, modulo two issues:

* I need to PR the links to the website, which will be done shortly
* Someone needs to write a top level intro to static guarantees

japaric, 2018-11-20:
> Someone needs to write a top level intro to static guarantees

Done in #92

closed 2018-11-20 by japaric