rust-embedded/book #55: Canonical link format

thejpster, 2018-10-19:
What should the canonical link format be?

HTML or Markdown?
```
[foo]: ./relative/markdown.md
[foo]: ./relative/html.html
```

Relative or absolute?
```
[foo]: ./relative/markdown.md
[foo]: /absolute/markdown.md
```

thejpster, 2018-10-20:
Take a look at https://github.com/thejpster/book/blob/testing_links/src/intro/install/verify.md.
It contains:

```
The contents may not match exactly but you should get the last line about
breakpoints and watchpoints. If you got it terminate the OpenOCD process and
move to the a [test1] b [test2] c [test3] d [test4] e [test5] f [test6].

[test1]: /intro/hardware.html
[test2]: ./intro/hardware.html
[test3]: /intro/hardware.md
[test4]: ./intro/hardware.md
[test5]: intro/hardware.md
[test6]: ../hardware.md
```

Only test1, test3 and test6 work in `mdbook serve`. Only test6 works on Github. Therefore I propose all URLs are made relative, with a `.md` extension, and no leading `./` where the file is in the same directory.

jamesmunns, 2018-10-30:
LGTM :+1: thanks for the investigation!

commit e310669d (2018-11-04): canonicalize URLs

as per #55

closes #55
commit e3140728 (2018-11-04): Merge #68

68: canonicalize URLs r=therealprof a=japaric

as per #55

closes #55

this PR also ignores print.html when `linkcheck`-ing as done in
rust-embedded/discovery#142
PR #68 (2018-11-04): canonicalize URLs
as per #55

closes #55

this PR also ignores print.html when `linkcheck`-ing as done in
rust-embedded/discovery#142
closed 2018-11-04 by bors[bot]