rust-embedded/book #65: Extend Portability Chapter

jamesmunns, 2018-10-30:
Moved from #36 

I would like to see the following introduced to the `embedded-hal`/portability chapter:

* Examples (we've already discussed)
* Commentary on why these traits are useful, (N*M vs N+M implementation, where N is # of chip/board support crates, and M is the # of driver traits).

Additionally, we may want something like [this image](https://screenshots.firefox.com/RviflD6enGaYiZRm/ferrous-systems.github.io), to demonstrate how this all fits together, and we may want to talk about the different components in the ecosystem somewhere:

* Peripheral Access Crates
* HAL Implementations
* Board Support Crates
* `embedded-hal`
* Driver Crates
* `no_std` libraries/crates (such as `heapless`, `menu`, etc.)

jamesmunns, 2018-11-20:
Addressed by #83 and #84, thanks @therealprof!

jamesmunns, 2018-11-20:
Still to do: Examples of each layer

TotalKrill, 2019-04-29:
After reading this book I still  feel like this chapter needs to be extended, I got a clear view of what the different crates shoul be (hal, PAC, drivers). But i do not feel that I have any clue on how to start utilizing these, or how to start creating or using drivers for embedded-hal, nor do i get any practical understanding of the benefits of rust ecosystem here. I actually felt more optimistic towards portability due to driver crates and embedded hal before reading this. 

I also feel I have no clear view of how to structure a project where I have to develop an embedded system using rust, and there are new devices that I need to fix a driver for.

Maybe some example on how to create a new embedded hal-crate for something existing on the stm32f3discovery, such as the acceleremoter, maybe a walk-through an example of how to get an embedded-hal trait object such as i2c from the discovery hal crate, create a basic driver for the accelerometer that accepts the i2c traits and is therefore portable. 

to tie everything together in the end, this could be extended to react on an interrupt event of the accelerometer, thus tying into the benefits of rust concurrancy etc.

korken89, 2019-04-29:
Ping @rust-embedded/resources