rust-embedded/book #97: Broken link in section 1.1 no_std

mone27, 2018-12-11:
At the bottom of the section 1.1 no_std there is a link for the FAQ, which is broken. It points to a page on rust-lang.org that does not longer exits and you get a 404 error.

shellwirt, 2019-01-31:
That's true, the FAQ actually doesn't seem to exist in the current rust-lang.org website. There's an open issue about this: rust-lang/www.rust-lang.org/issues/291

We could point to the old website FAQ instead: https://prev.rust-lang.org/en-US/faq.html#does-rust-work-without-the-standard-library

However, IMO the Summary does a good job describing what #![no_std] is. So I think it would be better to remove the link altogether.

nvll, 2019-03-10:
I ran into this today and spent a few minutes trying to find any sort of FAQ on the website. Since there doesn't seem to be one anymore, removing the link makes sense to me.