rust-embedded/book #118: Mac-specific installation doesn't work

harrisonturton, 2019-01-25:
Hello,

On [this page](https://rust-embedded.github.io/book/intro/install/macos.html), it asks us to run the following commands:

```
$ # GDB
$ brew cask install gcc-arm-embedded

$ brew install openocd

$ brew install qemu
```

Unfortunately, I get the error when executing the first command:

```
Error: Cask 'gcc-arm-embedded' is unavailable: No Cask with this name exists.
```

And when trying to tap `Caskroom/tap`:

```
htmacbook:local harrisonturton$ brew tap Caskroom/tap
==> Tapping caskroom/tap
Cloning into '/usr/local/Homebrew/Library/Taps/caskroom/homebrew-tap'...
remote: Repository not found.
fatal: repository 'https://github.com/Caskroom/homebrew-tap/' not found
Error: Failure while executing; `git clone https://github.com/Caskroom/homebrew-tap /usr/local/Homebrew/Library/Taps/caskroom/homebrew-tap --depth=1` exited with 128.
```

Any ideas? Not sure whats wrong.

ricjac, 2019-01-25:
Also noticed that caskroom is now deprecated (see github page): https://github.com/Caskroom/caskroom.github.io

Also, `gcc-arm-embedded` has been removed.

Go to the bottom of this gist, other ppl have been having the same problem:
https://gist.github.com/joegoggins/7763637

Look like you have two option:

1) 
brew tap PX4/px4
brew search px4
[install the latest px4/px4/gcc-arm-none-eabi-XX]

2)
Install `gcc-arm-none-eabi` yourself: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads/8-2018-q4-major
[Add the bin folder to your path - instructions in https://gist.github.com/joegoggins/7763637]

therealprof, 2019-01-25:
I'll take care of it.

commit 3e9429de (2019-01-26): Update macOS installation instructions

Fixes #118
commit be55c527 (2019-01-26): Merge #119

119: Update macOS installation instructions r=korken89 a=therealprof

Fixes #118
PR #119 (2019-01-26): Update macOS installation instructions
Fixes #118
closed 2019-01-26 by bors[bot]