rust-embedded/book #130: Gray colored text on book pages is not accessible to me

johnswolter, 2019-02-13:
The latest website page craze is to set the color of the text as gray.  No doubt there is a long-winded explanation from page designers.  Maybe it's the font choice.  No matter how you "view" it, It's a FAIL.   Maybe the font imposed this or some CSS was dropped onto the markup.  I'm trying to avoiding Monkey-ing around with the DOMs.
See or not see here: https://github.com/rust-embedded/wg

Some people, like myself, need the black text on an almost white background to read the text.  Sometimes the managers need to step-forward & assert design standards to web page developers.  

Add a way to change the page text colors using the font change choice I've seen at the top of some website pages.  This feature could also help when programmers have spent too much continuous time programming.  That is, they are just plain tired.

ryankurte, 2019-02-14:
Hey, thanks for the issue, it's a really good point!

We use [mdBook](https://github.com/rust-lang-nursery/mdBook) for rending all our resources. While we could look into adding a more accessible theme just for rust-embedded, this issue may be better placed on their repository so it can be fixed across all rust documentation?

As a sidenote, there is a theme selector in the top left of the mdBook window (see the following screenshot). None of them are particularly high contrast, but, maybe there is one that is better than the default until this is resolved.

[image]

ryankurte, 2019-03-13:
closing this for inactivity, feel free to re-open it if you have further thoughts.

closed 2019-03-13 by ryankurte