rust-embedded/book #132: epub/mobi/pdf version

asdfhjkl, 2019-02-15:
First of all, thanks to all contributors for your great effort!

I wonder if I'm overlooking something but is there no downloadable version of this book? If not, how hard is it to compile a pdf/epub/mobi myself?

adamgreen, 2019-02-15:
You might be able to click the printer icon in the upper right hand corner to print the document and then save it to a PDF instead of actually sending it to a printer. This works for me on macOS but it doesn't produce a ToC, just the pages of content.

I see that there is an [open issue](https://github.com/rust-lang-nursery/mdBook/issues/88#issuecomment-352153390) in the mdBook project for this.

asdfhjkl, 2019-05-16:
I'm sorry for the late reply. Your suggestion works.
I actually tried that before opening this issue but for some reason I got the impression that only the first page would be printed, which is wrong.

ildar, 2019-10-29:
what about other e-book formats? epub, fb2 ?
please reopen

ildar, 2020-10-13:
@asdfhjkl  ?

asdfhjkl, 2020-10-20:
I don't know what the proper etiquette is for reopening issues, so let me say sorry in advance if this irks somebody.
But since I apparently have the ability to reopen the issue and ildars question seems valid, I'll reopen it.

ryankurte, 2020-10-20:
hey @ildar, @asdfhjkl, as far as i know nothing much has changed. we continue to use `mdBook` to generate the docs outputs, the open issue for epub/other outputs via `mdBook` remains https://github.com/rust-lang/mdBook/issues/88, and the top right corner print -> to pdf option is still about the easiest approach as far as i know.

there is an [mdbook-epub](https://github.com/Michael-F-Bryan/mdbook-epub) plugin that may be worth a look (and [mdbook-latex](https://github.com/lbeckman314/mdbook-latex) though from the docs this appears to immature just yet), if the former works convincingly we could probably add this to our CI infrastructure (and if you were up to trying it out and opening a PR if it works well enough that would be excellent).

with respect to multiple ebook formats, i think it is unlikely that we will opt to generate more than one ebook type, you may be better served by an app like [calibre-ebook](https://calibre-ebook.com/) to convert between formats.

ildar, 2020-10-21:
pdf: I suggest that adding the link to Print → to PDF to the 1st chapter
would satisfy people.
epub: Again, a link in the 1st chapter to the "EPUB" file would be ideal.
If it's half-baked, then it's more or less ok also.

ildar, 2021-03-26:
I'd like to see the ebook/epub links for all the books on this topic.
Especially on this page: https://docs.rust-embedded.org/