rust-embedded/book #138: verify installation with qemu only

flip111, 2019-02-17:
https://rust-embedded.github.io/book/intro/tooling.html#qemu-system-arm

`Thanks to this you can follow some parts of this book even if you don't have any hardware with you!`

It's not clear what to do here https://rust-embedded.github.io/book/intro/install/verify.html
when just following the book with qemu only.

Suggestion 1:
`Skip this section when you don't have hardware`

Suggestion 2:
provide some commands to check the tooling without hardware

flip111, 2019-02-18:
i can't move this issue forward on my own at the moment

Heavypilgrim, 2023-10-25:
Hi @flip111,
I first thought so too.
But after continuing of reading the next chapters, the book explains the way how to simulate and debug the code with qemu (see chapter "2.1 qemu" https://doc.rust-lang.org/stable/embedded-book/start/qemu.html). Only the chapter "2.2 Hardware https://doc.rust-lang.org/stable/embedded-book/start/hardware.html" isn't possible without proper hardware.

So have fun with the embedded rust stuff.

Cheers,
Heavy