rust-embedded/book #146: $TRIPLE, does it have any meaning ??

flip111, 2019-02-17:
https://rust-embedded.github.io/book/start/qemu.html#cross-compiling

flip111, 2019-02-18:
i can't move this issue forward on my own at the moment

adamgreen, 2019-02-19:
It appears to be a variable placeholder for whatever target triple you want to build. A little later it shows an example triple of `thumbv7m-none-eabi` and how this was set as the default target triple in the .cargo/config file.

flip111, 2019-02-19:
what's a "target triple" ?

caemor, 2019-02-19:
Here is a longer description: https://clang.llvm.org/docs/CrossCompilation.html#target-triple

It's basically the target that you want to compile for and it's seperated in 3 parts. Therefore target triple.

flip111, 2019-02-19:
Ok i will wait on some other PR's before reconsidering this.