rust-embedded/book #154: Add chapter about async programming

flip111, 2019-02-17:


flip111, 2019-02-18:
Some resources:

Allocators that don't need std
https://github.com/Nemo157/embrio-rs/
https://josh.robsonchase.com/embedded-executor-2/
https://github.com/japaric/no-std-async-experiments
taken from: https://github.com/rust-embedded/wg/issues/23

Interesting read: http://blog.japaric.io/brave-new-io/#async-hal

https://www.bobbin.io/

https://docs.rs/embedded-hal/0.2.2/embedded_hal/

SergioGasquez, 2023-12-11:
I think this would be very helpful to have, in my opinion, maybe it should not explain the details of async, for that we can just point to https://rust-lang.github.io/async-book/ or any other resource, but maybe we could explain frameworks like [RTIC](https://rtic.rs/2/book/en/) or [Embassy](https://embassy.dev/)