rust-embedded/book #157: The C approach

flip111, 2019-02-17:
https://rust-embedded.github.io/book/peripherals/a-first-attempt.html#the-c-approach

"The C approach" -> title reads as if i would get to read some C code how to work with peripherals. In this context it's only a "coincidence" (it's not but ok), that we have to use the `#[repr(C)]` pragma to keep the memory layout (which is the point of this section).

Suggestion:
rename to "Address map and Rust structs"

Also have stronger connection in the description how hexidecimals from an address are mapped to structs in rust. Now it's just strange that we are going from a description of an address space to a rust struct immediately without having some code (in rust) that shows this mapping.

flip111, 2019-02-17:
Ah i understand now why the naming of the section once i start reading the next section "Now, there are a couple of problems with the approach above.". Even so the title of a section should relate to the content of the section. To keep the emphasize on the difference between Rust and C, the "Volatile Accesses" can start describing this

suggestion:
Just using plain structs is the only thing possible in C (without adding extra program code). In rust however it's possible to provide more safety. <.. refer to concept of zero abstractions ..>.

I'm wondering, is it true though? C also has macros.

flip111, 2019-02-17:
Didn't want to open another issue for this. About https://rust-embedded.github.io/book/peripherals/a-first-attempt.html#volatile-accesses

> Now, the problem is that compilers are clever.

This is strange to read. First the write lays out 4 problems. Then continues with "the problem is ..."

> Most importantly, it doesn't actually work...

This is not good. This reader has just spend time trying to grasp some concepts and now you tell them it's invalid. This is frustrating. Instead section "The C Approach" should explain "if we would write code in rust like we would in C, it would look like this. However this is not possible in rust as we will explain later".

> the compiler can notice this and just skip the first write entirely

> So, we've fixed one of our four problems

The problem of the compiler skipping statements was never mentioned in the 4 mentioned problems. To which of the 4 mentioned problems is "we've fixed one of the four" referring to ??

---

Use bullet-points or numbered lists when it's important for the reader to remember things. This list of  4 problems is just fluff now. This can also be noticed that these 4 problems are nowhere addressed later in the text. Remove this list and replace it with a single sentence.

flip111, 2019-02-17:
> As the driver author

This reads strange because "driver" can also be a person. suggestion: as the author of the code (a driver) controller this hardware.

It's longer but it connects better to the previous sections of the text. Now we gradually introduced the name of "driver" keeping the connection with other text.

flip111, 2019-02-17:
Didn't want to create a separate issue for this one either

https://rust-embedded.github.io/book/peripherals/a-first-attempt.html#the-rusty-wrapper

> We need to wrap this struct up into a higher-layer API that is safe for our users to call

> Now, the problem with this approach is that the following code is perfectly acceptable to the compiler

Please don't waste the readers time coming up with some code that has a problem and not even provide a solution. What is the point of this entire section anyway ??

andre-richter, 2019-02-17:
Hi @flip111 ,

thank you for your thorough review of the book, and also the pull requests you sent so far. We value your input.

Please let me kindly remind you that the authors of the book and other content of rust-embedded are doing their work mostly in their free time, because they believe in the language and they want to push its popularity and encourage learning it.
A part of encouraging people to spent their free time on writing such resources is a welcoming community, and tone.
Some of your open issues so far come across with a commanding tone, which is is not encouraging, and also not really appropriate because this is a free working group and no one owes anyone a service.

We are really keen to improve our documentation based on your feedback, but please help us having fun while doing so.

Thanks,
Andre

flip111, 2019-02-17:
Hi sorry for the tone. It's been already a lot of work even doing the review. I didn't put the extra effort in thinking about how it would come across.

This is entirely unrelated how i feel about the book. I think it's really amazing things got so far and that there is actually a book now for embedded programming of rust.

Also i like to inform you that i didn't look at who wrote this, so i'm not addressing anyone personally. Though i can not help someone might feel offended when i mention the passage written by that person.

Also from my side it's voluntary. I simply don't want to put the additional effort now to think about my own wording. If this is not acceptable for the code of conduct or some other principle i will just stop my review for the moment and you can delete the created issues as well if you wish to do so.