rust-embedded/book #161: Hardware and rust

flip111, 2019-02-17:
In the hardware section early on in the book it would be really nice to go over some of the hardware difference that come up while programming with rust.

I read later

https://rust-embedded.github.io/book/concurrency/index.html#atomic-access

> Specifically for Cortex-M, thumbv6 (Cortex-M0) does not provide atomic instructions, while thumbv7 (Cortex-M3 and above) do. These instructions give an alternative to the heavy-handed disabling of all interrupts:

flip111, 2019-02-18:
I will wait for the close or merge of https://github.com/rust-embedded/book/pull/169 before re-considering this.

stappersg, 2023-03-11:
> I will wait for the close or merge of #169 before re-considering this.

#169 has been merged.  @flip111, you can now reconsider this issue.

flip111, 2023-03-11:
I don't want to work on this at this moment, i will close it.

closed 2023-03-11 by flip111