rust-embedded/book #181: STM32F103CB openocd issue

nifker, 2019-03-30:
upon using "openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg" I get:
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: open failed
in procedure 'init' 
in procedure 'ocd_bouncer'

therealprof, 2019-03-30:
You need to use a target configuration for the f1, not the f3.

tim-seoss, 2019-03-30:
I use this `openocd.cfg`:

```
# OpenOCD configuration for the ("bluepill") STM32F103C8 development board

# Depending on the external ST Link programmer you're using, you'll have to pick ONE
# of these interfaces. At any time only one interface should be commented out.

# External v2.1 stlink
#source [find interface/stlink-v2-1.cfg]

# External v2.0 stlink
source [find interface/stlink-v2.cfg]

source [find target/stm32f1x.cfg]
```

Other gotchas with these boards are:

- You will need to hold down reset whilst starting openocd if the board has previously had code flashed which puts the code into a low power state (because that disables the one wire debug interface).
- You may need to unlock the flash before it can be written, using this command at the gdb prompt:

```
(gdb) monitor stm32f1x unlock 0
```

nifker, 2019-03-30:
if I use "openocd -f interface/stlink-v2-1.cfg -f target/stm32f1x.cfg" I get the same output

therealprof, 2019-03-30:
@nifker Is the device properly hooked up?

nifker, 2019-03-30:
@therealprof
quite sure about it.
3.3V on pin 8
swclk on pin 6
gnd on pin 4
swio on pin 2

nifker, 2019-03-30:
@tim-seoss had the proper solution. thanks

closed 2019-03-30 by nifker

huberemanuel, 2019-10-23:
What was the solution?

nifker, 2019-10-24:
@huberemanuel it was about the ST-Link version in the config