rust-embedded/book #189: Not seeing the device in lsusb output

amanjeev, 2019-05-11:
Hello, absolutely new to the embedded world. My issue is that when I plug in the device to my laptop (running Ubuntu 19.04), I do not see the device in the list produced by `lsusb`.

The laptop I have has USB3 ports only and I use a USB2 converter/adapters in the middle. I have tried three such adapters so maybe it is not the adapters. I see the lights flashing on the board so it is powered up I guess.

I wonder if anyone has tried this with USB3 ports.

ryankurte, 2019-05-11:
hi hi, if you're getting USB errors you might be able to see what's going on using `sudo dmesg | grep usb`.

it might also be worth checking the device works with another computer, but other than that it's pretty hard to debug remotely and usb-c to usb adaptors and cables are pretty cursed  :-/

amanjeev, 2019-05-12:
Oh yea! I am sorry to ask this stupid question. OTOH now I think I may have to recompile the kernel or something. :laughing:

closed 2019-05-12 by amanjeev

amanjeev, 2019-05-12:
Just in case someone was curious, the issue was with my USB cable. Bought a new USB mini to USB cable and the output `lsusb` shows something like 

```Bus 001 Device 005: ID 0483:374b STMicroelectronics ST-LINK/V2.1```

Even though with the older faulty cable, the LEDs were blinking there was clearly an issue. Went to the nearest Microcenter and got a new cable. :heart:

therealprof, 2019-05-12:
@amanjeev Could also have been a "power-only" cable, those darn things still exist unfortunately. Great that you got it to work, though!