rust-embedded/book #190: `debug::exit()` is commented out but mentioned as executed

amanjeev, 2019-05-12:
[The advice here is that](https://github.com/rust-embedded/book/blame/master/src/start/hardware.md#L82) 
> Make sure the `debug::exit()` call is commented out or removed

However, [in the same document below under](https://github.com/rust-embedded/book/blame/master/src/start/hardware.md#L249) `gdb` 
> Issuing another `next` will make the processor execute `debug::exit`

Am I mistaken in my understanding that there won't be any execution of `debug::exit()` because it is commented out? Also, that it is hang forever after the second ```(gdb) next```?

I could update the document with a PR if you can please advise me.

stefaneicher, 2019-07-15:
I saw the same issue. Would be helpful to make the documentation consistent according to the description of amanjeev.

leshow, 2019-07-23:
Sorry to respond to this random thread, but looking at past issues it looks like some aren't being addressed. Just curious if either of you were actually able to get output on the board? Mine just loops over `Info : halted: PC: 0x0800184a` indefinitely without ever printing "hello world"

mirao, 2022-07-05:
I noticed the same issue with inconsistency in docs as originally reported by @amanjeev .
I'm testing it with MCU STM32F407 (former STM32F4DISCOVERY) with adjusted `memory.x`. I'm using `Debug (OpenOCD)` in VSCode.

As for functionality, "Hello world!" is printed even with `debug::exit` in code and finally application "freezes" in `loop {}` forever.

Output with `debug::exit(debug::EXIT_SUCCESS);`:
```
Hello, world!
semihosting: *** application exited normally ***
target halted due to breakpoint, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000b7a msp: 0x2001ffa8, semihosting
```

Without `debug::exit(debug::EXIT_SUCCESS);`:
```
Hello, world!
```

eldruin, 2022-07-25:
Fixed in #323. Thanks!

closed 2022-07-25 by eldruin