rust-embedded/book #192: Gdb stuck on "Continuing."

gmantaos, 2019-05-20:
Hello,

First of all thank you for the work on this project, the book has been very helpful so far.
So I've followed all the steps so far, but any attempts at debugging or running the program after `load` just get stuck forever at `Continuing.` on GDB, like in the screenshot below.

[image]

My board is a **Nucleo-L4R5ZI** with an `STM32L4R5ZIT6U`.

The program is basically just the hello example.

```rust
#[entry]
fn main() -> ! 
{
    asm::nop();
    
    hprintln!("Hello, world!").unwrap();
	
    loop
    {
    }
}
```

This is actually my embedded project, so any help towards what I should investigate would be very helpful :)

leshow, 2019-07-23:
I got think also at one point, check to make sure you have the right `memory.x` for your board

gmantaos, 2019-08-07:
> I got think also at one point, check to make sure you have the right `memory.x` for your board

Ye that was it, I was reading the datasheet wrong.
Sorry I forgot to close the issue!

closed 2019-08-07 by gmantaos

victor-wei126, 2022-10-22:
What is `memory.x`? I'm running into the same problem.

victor-wei126, 2022-12-23:
@dilnozacs yeah it turns out my issue iirc was that I was working in the wrong directory I think the `f3discovery/` and not `microbit`. so the `memory.x` file was wrong so you might wanna check that.

dilnozacs, 2022-12-24:
@victor-wei126 can you elaborate? my `memory.x` file currently is located under `microbit/src/05-led-roulette/memory.x` path with the following content:
```
MEMORY
{
  /* NOTE K = KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x00000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 16K
}
```
I never touched the file so I'm not sure what could be "changed"? P.S I am running the gdp command while within `rust/discovery/microbit` directory

dilnozacs, 2022-12-24:
please disregard it, weirdly enough rerunning everything, it finally picked up the target file and is pointing now in the right breakpoint entry and exiting `continue` as expected. Thank you for responding to the earlier comment! 😸

victor-wei126, 2022-12-24:
Np and glad you figured it out :). fwiw, I think you were in the right directory anyways