rust-embedded/book #199: Output doesn't match book STM32F303

leshow, 2019-07-22:
I have been following the guide up to [hardware](https://docs.rust-embedded.org/book/start/hardware.html). I have the stmf32f303 exactly the same board as describe in the book. When I plugged it in before the board used to light up and I could press the 'user' button to cycle through the accelerometer and gyroscope. Now the board just has a red power indicator and the COM set to red all the time.

I had followed the guide up to running `hello.rs`, leaving openocd open, and connecting w/ gdb.

from openocd:
```
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v15 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.913064
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x10036422
Info : flash size = 256kbytes
undefined debug reason 7 - target needs reset
semihosting is enabled
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
adapter speed: 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080015ca msp: 0x20010000, semihosting
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
Info : Padding image section 0 with 10 bytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20010000, semihosting
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
adapter speed: 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080015ca msp: 0x20010000, semihosting
Info : halted: PC: 0x0800163a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
Info : halted: PC: 0x0800184a
```

and from gdb:

```
Transfer rate: 12 KB/sec, 2403 bytes/write.
Note: automatically using hardware breakpoints for read-only addresses.
DefaultPreInit () at /home/leshow/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.9/src/lib.rs:564
564     pub unsafe extern "C" fn DefaultPreInit() {}
(gdb) break main
Note: breakpoint 4 also set at pc 0x8000480.
Breakpoint 5 at 0x8000480: file examples/hello.rs, line 13.
(gdb) continue
Continuing.

Breakpoint 2, HardFault_ (ef=0x0)
    at /home/leshow/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.9/src/lib.rs:548
548             atomic::compiler_fence(Ordering::SeqCst);
(gdb) next
^C
Program received signal SIGINT, Interrupt.
0x0800184a in HardFault_ (ef=0x0)
    at /home/leshow/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.9/src/lib.rs:548
548             atomic::compiler_fence(Ordering::SeqCst);
```

Also, my `memory.x` file had ` FLASH : ORIGIN = 0x00000000, LENGTH = 256K` in it, but the book said `0x08000000` so I've changed it to that. I'm not sure if that's the correct value.

leshow, 2019-07-23:
I started messing around with the [f3](https://github.com/japaric/f3) project trying to get the examples running, I was able to get the `hello.rs` in the examples and `blinky.rs` running with gdb on my board. 

I noticed the `memory.x` files are different, so it's contents into my cortex-m-quickstart clone and voila. I was able to debug / run.

```
MEMORY
{
  CCRAM : ORIGIN = 0x10000000, LENGTH = 8K
  FLASH : ORIGIN = 0x08000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
}

_stack_start = ORIGIN(CCRAM) + LENGTH(CCRAM)
```

These were the values that fixed it for me, I'll submit an issue to the quickstart

closed 2019-09-07 by leshow

crazycarl, 2019-11-09:
This just helped get me out of a rut. Thanks for this.
Since the book states that its meant to be used with an STM32F3DISCOVERY, it really should assume you're using an STM32F3DISCOVERY.