rust-embedded/book #201: "Architecture rejected target-supplied description" when trying to connect to STM32F3 using GDB

johnwstanford, 2019-07-29:
I'm working through section 2.2 and OpenOCD seems to connect to the device fine, but when I try `target remote :3333` in GDB, I get:

```
warning: Architecture rejected target-supplied description
Truncated register 16 in remote 'g' packet
```

If I ignore the warning and try `load`, I get `You can't do that when your target is exec`.

I was able to verify the hardware as described in section 1.4.4 (Verify Installation).  I'm on Ubuntu 18.04.

adamgreig, 2019-07-29:
Are you running `gdb` or `arm-none-eabi-gdb` or `gdb-multiarch`?

johnwstanford, 2019-07-29:
It was `gdb`.  It worked when I tried `gdb-multiarch`.  Thanks!

I saw the brackets around `gdb` in the text and I wasn't sure what they meant.  I guess they must mean that there are a few different possibilities for GDB.  I may have missed that if it was explained earlier.

closed 2019-07-29 by johnwstanford

adamgreig, 2019-07-29:
I think the brackets are just meant to indicate the default gdb shell prompt, which is `(gdb)`. In general you always have to use gdb (and gcc, ld, objdump, etc) for your specific architecture, which is why the ARM gcc embedded package includes `arm-none-eabi-gdb`. However new gdb has a `gdb-multiarch` option which seems fine too. Glad you've got it working.