rust-embedded/book #204: OpenOCD does not work on USB3.0 ports on Windows

hlavaatch, 2019-08-10:
I had a hard time trying to follow the hardware setup chapter on Windows 10.

OpenOCD would just not see the STM32F3DISCOVERY board even though it shows up fine on device manager and works fine with the ST software.

It turns out USB 3.0 capable ports do not work with OpenOCD, it fails with "Error: open failed".

Please add a note in the book to use USB 2.0 USB port (which works) as a workaround until the issue with OpenOCD is (hopefully) resolved.

burrbull, 2019-08-10:
Can you add issue link here?

hlavaatch, 2019-08-12:
I don't have a proper issue link, OpenOCD guys don't take this seriously ;(
There are several reports of this in their bug database over the years, and they don't care... It may do with them using prehistoric version of libusb and not willing to update.

I will post a link if I pursue this in the future, meanwhile I'm more interested in writing embedded Rust than fixing openocd, obviously :)

hlavaatch, 2019-08-22:
Here is one [(https://sourceforge.net/p/openocd/tickets/249/)]

steveklabnik, 2020-06-22:
I just hit this today :(

ryankurte, 2020-06-23:
ooof, frustrating, and definitely something we should note or perhaps, move away from OpenOCD for this section.

As other alternatives, for ST boards (and the STM32F3DISCOVERY) you may also be able to use [stlink-org/stlink](https://github.com/stlink-org/stlink) or the official ST utils to interact with the existing debugger firmware or  flash the debugger with [jlink firmware](https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/) and use the JLink tools, either of which should be usable without openOCD.

hlavaatch, 2023-08-11:
I am happy to report that the latest version of openocd I got from https://xpack.github.io/blog/2023/01/30/openocd-v0-12-0-1-released/ seems to be working fine with USB3 port on Windows 11! I guess this issue can be closed now

closed 2023-08-11 by hlavaatch