rust-embedded/book #209: Verifying installation failed on MacOS

wusyong, 2019-11-06:
I followed the instructions to install tools needed on MacOS but failed with error like this:

```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v15 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.933492
Error: init mode failed (unable to connect to the target)
in procedure 'init'
in procedure 'ocd_bouncer'
```

I have tried steps book suggested and couldn't find solution either. Does anyone can help?

raiyan-machinemax, 2019-11-09:
I'm having the same issue. Here is my output.

```
Raiyans-MacBook-Pro:~ raiyan$ openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg
Open On-Chip Debugger 0.10.0+dev-gf6f4a675 (2018-08-06-10:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: open failed
in procedure 'init'
in procedure 'ocd_bouncer'

Raiyans-MacBook-Pro:~ raiyan$
```

I'm running it on MacBook Pro 2018 with USB-C Ports. Running MacOS X Catalina.

Just to mention I was using OpenOCD for work on this machine without any issue.

raiyan-machinemax, 2019-11-13:
The problem for me was with my USB cable. Replaced it and openOCD is working again.

zacck, 2020-10-01:
+1 on this issue I am however using a  STM32F103C8T6  aka Blue Pill and attempting to connect via the single microusb port on the device

kramerjul, 2021-02-11:
Thx @raiyan1982. Same for me. I changed the USB cable and now it works fine for me.

wusyong, 2021-02-12:
Close it because I also changed the cable to resolve it.

closed 2021-02-12 by wusyong