rust-embedded/book #234: Need help debugging qemu

fujexo, 2020-05-01:
I'm not sure if I am missing something or not. I think I don't have everything neccessary installed but cannot figure out what's missing and how to install it.

`src/libcore/slice/mod.rs: No such file or directory.`

Maybe you can point me in the right direction?

```
$ cargo build --target thumbv7m-none-eabi --example hello --release
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling proc-macro2 v1.0.10
   Compiling unicode-xid v0.2.0
   Compiling syn v1.0.18
   Compiling cortex-m v0.6.2
   Compiling stable_deref_trait v1.1.1
   Compiling vcell v0.1.2
   Compiling cortex-m-rt v0.6.12
   Compiling cortex-m-semihosting v0.3.5
   Compiling r0 v0.2.2
   Compiling panic-halt v0.2.0
   Compiling semver v0.9.0
   Compiling volatile-register v0.2.0
   Compiling rustc_version v0.2.3
   Compiling quote v1.0.4
   Compiling bare-metal v0.2.5
   Compiling generic-array v0.12.3
   Compiling generic-array v0.13.2
   Compiling as-slice v0.1.3
   Compiling aligned v0.3.2
   Compiling cortex-m-rt-macros v0.1.8
   Compiling app v0.1.0 (/home/fuj/git/home/rust-playground/embedded/app)
    Finished release [optimized + debuginfo] target(s) in 54.48s

$ qemu-system-arm \
                         -cpu cortex-m3 \
                         -machine lm3s6965evb \
                         -nographic \
                         -semihosting-config enable=on,target=native \
                         -gdb tcp::3333 \
                         -S \
                         -kernel target/thumbv7m-none-eabi/debug/examples/hello
Hello, world!
```

When I attach gdb, I get an error that some files are missing:

```
$ arm-none-eabi-gdb -f target/thumbv7m-none-eabi/debug/examples/hello
GNU gdb (GDB) 9.1
<snip>
Reading symbols from target/thumbv7m-none-eabi/debug/examples/hello...
(gdb) target remote :3333
Remote debugging using :3333
0x000012cc in <core::ops::range::Range<usize> as core::slice::SliceIndex<[T]>>::index_mut () at src/libcore/slice/mod.rs:2927
2927    src/libcore/slice/mod.rs: No such file or directory.
(gdb) break main
Breakpoint 1 at 0x444: file examples/hello.rs, line 11.
(gdb) continue
Continuing.

Breakpoint 1, main () at examples/hello.rs:11
/home/fuj/git/home/rust-playground/embedded/app/examples/hello.rs:11:186:beg:0x444
11      #[entry]
(gdb) next
[Inferior 1 (process 1) exited normally]
(gdb) quit
```

blelem, 2020-05-16:
I am facing the same problems.
I found some workarounds, but they are not great, I do hope that experts reading this can think of real solutions for these problems. 

The first issue, the "src/libcore/slice/mod.rs: No such file or directory", seems to be related to the rust linker, which I presume generates symbol file that is not understood properly understood by gdb. gdb fails to map the compiled code to the source code. I got that fixed by using the gnu linker. In the 
.cargo/config file, comment the line with LLD, and uncomment the line with GNU linker.
```
  # LLD (shipped with the Rust toolchain) is used as the default linker
  #"-C", "link-arg=-Tlink.x",  #<--- comment

  # if you run into problems with LLD switch to the GNU linker by commenting out
  # this line
  "-C", "linker=arm-none-eabi-ld", #<--- uncomment
```
When it comes to stepping through, `next` or `step` do not seem to work, gdb just runs till the end of the program. Breakpoints do seem to work reliably though, so as a poor's man way to step through the code, you can set a breakpoint on the next line, then `continue` until there.

blelem, 2020-05-16:
Pinging @japaric, since he worked on the .cargo/config file for the cortex-m-quickstart project.

blelem, 2020-05-17:
For the issue with the `next` not breaking on the next line, try to do a 
`break hello::__cortex_m_rt_main`
instead of a 
`break main`

After hitting that breakpoint, stepping through the code works normally.
I've proposed changes to the book via this PR #239

commit 1e6247eb (2020-05-24): Merge #239

239: QEMU debugging updates r=adamgreig a=blelem

Addressing a few issues:
Fixes #237, Fixes #234, Fixes #217 

The biggest issue this PR is addressing is correcting the "debugging QEMU with GDB" section that really didn't work with latest ARM toolchain (on my setup), because of two things:
 - gdb didn't map the rust libcore functions properly.  For example, after connecting to a QEMU instance, it did not show the current PC to be on the Reset function, but some random method.  
 - gdb didn't step over (`next`), after breaking at `main`. It just ran to the end of the program.

The  [... cut]
PR #239 (2020-05-24): QEMU debugging updates
Addressing a few issues:
Fixes #237, Fixes #234, Fixes #217 

The biggest issue this PR is addressing is correcting the "debugging QEMU with GDB" section that really didn't work with latest ARM toolchain (on my setup), because of two things:
 - gdb didn't map the rust libcore functions properly.  For example, after connecting to a QEMU instance, it did not show the current PC to be on the Reset function, but some random method.  
 - gdb didn't step over (`next`), after breaking at `main`. It just ran to the end of the program.

The user is now instructed to disable the L [... cut]
closed 2020-05-24 by bors[bot]