rust-embedded/book #244: Error: Can't find interface/stlink.cfg in procedure 'script'

connorjcantrell, 2020-05-30:
After following the MacOS [installation instructions](https://rust-embedded.github.io/book/intro/install/macos.html), I am having an issue verifying my installation.

I am connected to "USB ST-LINK", and my ST-LINK header is populated

```
$ openocd -f interface/stlink.cfg -f target/stm32f3x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
```

therealprof, 2020-05-30:
@connorjcantrell In this stale version of OpenOCD the config file is still called interfaces/stlink-v2.cfg (or stlink-v2-1.cfg). In the current development version using those files will cause a warning and maybe in the next release it will be gone but who knows when that will arive, still a few years left before the 10th anniversary of 0.10.0.

therealprof, 2020-05-30:
I would recommend checking out https://probe.rs/ instead.

connorjcantrell, 2020-05-30:
@therealprof thank you, `stlink-v2-1.cfg` worked.

Thanks for the recommendation, OpenOCD has been problematic in so many ways. I will give probe.rs a try

closed 2020-05-30 by connorjcantrell