rust-embedded/book #249: Fedora has no package named "arm-none-eabi-gdb"

Qwertie-, 2020-06-14:
The install chapter for linux advises to install this package but it does not exist on fedora 32 or at least not under that name.

adamgreig, 2020-06-14:
Perhaps `gdb-multiarch` is available?

Qwertie-, 2020-06-15:
It is not. Is it possible that it could have been bundled in with the main gdb package? I have been following the book and I don't seem to have had any issues while not having that package.

ryankurte, 2020-06-15:
not at all familiar with fedora packaging but it looks like it was dropped due to a failed build / missing python dependency?

https://bugzilla.redhat.com/show_bug.cgi?id=1603406

jcrossley3, 2020-08-25:
Any suggestions for a workaround to follow the book on Fedora 30 or higher? Thanks!

ryankurte, 2020-08-26:
it appears arm now package arm-none-eabi-gdb in their binary distributions available [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads), if this works for you we could update the book to offer this alternative.

aneonlight, 2022-01-25:
It seems that this has yet to be fixed in Fedora 35.

I can try and scrape together a pull request. Although, it would break away from the package manager so I'm unsure if a general install vs Fedora specific would be better. (I can do both, just looking for a little guidance)

tylerferrara, 2022-10-30:
I've been struggling with this and found this process to be the best method forward:

Visit the download page for the latest *arm-gnu-toolchain* -> https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

```fish
export ARM_TOOLCHAIN="your-arm-gnu-toolchain-url"

# Download and extract to /opt
wget -qO- ${ARM_TOOLCHAIN} | tar xv -C /opt -J --transform 's!^[^/]*!arm-none-eabi!'

# Add binaries to path
echo 'export PATH="/opt/arm-none-eabi/bin:$PATH"' >> ~/.bashrc
source ~/.bashrc

# Install ncurses-libs
sudo dnf install ncurses-libs -y

# Create required libraries from latest version
sudo cp -rf /usr/lib64/libncurses.so /usr/lib64/libncurses.so.5
sudo cp -rf /usr/lib64/libncursesw.so /usr/lib64/libtinfo.so.5
```

Let me know if this helps anyone!

mavjs, 2022-11-14:
I am on Fedora 37 and just started following the book. I also noticed that `arm-none-eabi-gdb` has be unmaintained and thus considered dead. (Ref: https://src.fedoraproject.org/rpms/arm-none-eabi-gdb )

I have gdb version `GNU gdb (GDB) Fedora Linux 12.1-6.fc37`, so far I have gotten to https://docs.rust-embedded.org/book/start/hardware.html and have just used that `gdb`, and all the steps are working.

I took a look at `gdb-multiarch` PKGBUILD from ArchLinux - https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=gdb-multiarch#n27 and noticed that it has configure option of `--enable-targets=all`

From there, I tried to check what targets Fedora have enabled then, and noticed: `--enable-targets=s390-linux-gnu,powerpc-linux-gnu,arm-linux-gnu,aarch64-linux-gnu` in https://src.fedoraproject.org/rpms/gdb/blob/f37/f/gdb.spec#_645

I do not know a lot about the differences with `arm-none-eabi-gdb`, however, it seems to me we may not not it anymore(?) So perhaps, someone could confirm this is the case for them as well, and I could look at sending PR to fix Fedora parts of the installation steps in the book.

[image]

closed 2022-11-17 by bors[bot]