rust-embedded/book #263: New Discovery boards use a stlink-v2-1 and need different openocd config file

Phylex, 2020-09-05:
Hi in the "Verify Installation" section of the book openocd is started with the interface/stlink.cfg or the /stlink-v2.cfg new boards use a stlink-v2-1 so the correct file is the interface/stlink-v2-1.cfg then it starts without a problem otherwise it errors with
``
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: open failed
in procedure 'init' 
in procedure 'ocd_bouncer'
``

adamgreig, 2020-09-05:
Thanks for the report! What version of openocd are you using? In the latest version, there's no `stlink-v2.cfg` or `stlink-v2-1.cfg`, and all STlinks use just `stlink.cfg` instead. However, there hasn't been a release of openocd in a long time, so many users are still on old versions where those other files are required.

Probably the complete solution here is that we'll stop suggesting openocd and suggest probe-rs tools instead, which don't have this sort of issue.

Phylex, 2020-09-16:
The version is 0.10.0 and is the current packaged version for void-linux.

bitdivine, 2020-09-28:
Same symptoms, solution and openocd version here.  0.10.0 is also the default version of openocd for Ububtu 20.04.  I note that for programming Blue Pills I also use stlink-v2-1.cfg

julianoes, 2020-10-17:
This issue seems to have been fixed by #268, however, it would be good to add to the troubleshooting section to try both, if one doesn't work. The F3 I bought seemed to still want `v2`, not `v2-1`.

jeffwhite-619, 2020-11-21:
I have recently ordered a Discovery board for use in this tutorial and had to change the filename to stlink-v2-1.cfg for it to work. I'm not sure if I received an older board, or if this is the newer board, but it does appear that users like myself may be still be acquiring boards that need this change to proceed through the tutorial.