rust-embedded/book #264: Rust on the STM32F407VET6 using ST-LINK

coreychristian, 2020-09-10:
Hi, is it possible to get some help setting up my board (linked below)

https://os.mbed.com/users/hudakz/code/STM32F407VET6_Hello/shortlog/

The issue I have is related to connecting to the board to start flashing or debugging code. I have some experience with boards like arduino, and ESP8266 running micropython etc, and would like to get  my hands dirty with some STM32 stuff.

I reached the stage where I am running the openocd -f interface/stlink-v2.cfg -f target/stm32f3x.cfg command, but openocd doesnt seem to be connecting to the board. I have managed to connect to the board using USART and one of the ST applications, but for some reason cannot connect using SWD. 
I get the following 

For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Error: init failed
in procedure 'init' 
in procedure 'ocd_bouncer'

I have both boot pins on the board connected to ground, but have tried all the different configurations with no joy. Am  I missing something?

coreychristian, 2020-09-10:
I have also tried the openocd command using the -f target/stm32f4x.cfg flag.

therealprof, 2020-09-10:
Hi there. I'm afraid the book is not _quite_ up-to-date with the latest and greatest happening in the Rust ecosystem.

The latest and greatest achievements Re: flashing and debugging are https://probe.rs and https://ferrous-systems.com/blog/probe-run/

Re this particular board you might want to have a look at https://github.com/stm32-rs/stm32f4xx-hal, there're also examples which should run just fine

Last but not least we have always people happy to help with STM32 speicific problems on https://app.element.io/#/room/#stm32-rs:matrix.org

coreychristian, 2020-09-12:
Thanks, have tried to setup using probe-rs-cli. As I am new to SWD, I am not sure where I am going wrong, but I am beginning to think I'm going wrong somewhere at the hardware level, connecting the ST-Link (clone) and trying to run RUST_LOG=probe_rs=debug probe-rs-cli info I have noticed that the first time I run it, the ST-Link seems to be in DFU mode, then it leaves this mode and the only way it returns is by unplugging it and plugging it back in again. (See log below)
 
DEBUG probe_rs::probe::daplink::tools > Attempting to open 0483:3748 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 39, bNumInterfaces: 1, bConfigurationValue: 1, iConfiguration: 0, bmAttributes: 128, bMaxPower: 50, extra: None }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 0, bDeviceSubClass: 0, bDeviceProtocol: 0, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14152, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: Dfu
 DEBUG probe_rs::probe::stlink                > STLink version: (2, 32)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 WARN  probe_rs::probe::stlink                > check_status failed: JtagGetIdcodeError
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
Error: An error with the usage of the probe occured

Caused by:
    0: An error specific to a probe type occured
    1: Command failed with status JtagGetIdcodeError

I have connected the board to the ST-Link using four wires as documented online, clk -> clk, swdio -> swdio and vcc and ground, also tried all combinations for the boot 0 and 1 pins, bringing them to board voltage and ground, no luck. Any help will be much appreciated.

therealprof, 2020-09-13:
Weird, haven't seen that output before. For debugging there's no need to muck around with the boot pins. What's the firmware on your ST-Link clone? Mind joining us on element? It's much easier to debug things live than than through an GH issue...

coreychristian, 2020-09-14:
Closing because this was a hardware related issue rather, am unable to say exactly what the issue was but, I tried again today after reading something about the reset button needing to be pressed on some boards, before I even reached that point, I saw the led's on my ST-Link clone blink in a different pattern, and sure enough I was in the money :)

closed 2020-09-14 by coreychristian