rust-embedded/book #267: Arch Linux: `target remote :3333` "Connection timed out"; needs IP address

michd, 2020-10-02:
In [2.2 Hardware](https://doc.rust-lang.org/stable/embedded-book/start/hardware.html), 

```
(gdb) target remote :3333
```
is listed to connect gdb to openocd. When I ran this on my machine, it consistently timed out:

```
:3333: Connection timed out.
```
The command I'm invoking for the gbd console is:

`arm-none-eabi-gdb -q target/thumbv7em-none-eabihf/debug/examples/hello`

Despite openocd indeed running, according to `netstat -tulpn | grep LISTEN`:
```
tcp        0      0 0.0.0.0:3333            0.0.0.0:*               LISTEN      13303/openocd
```

This gave me the idea to try the following:

```
(gdp) target remote 0.0.0.0:3333
```

This provided the result described in the book. Perhaps it would help some if this solution was added as a note in the book.

cr0ck3t, 2023-07-20:
I ran into a similar problem with `netstat -tulpn | grep LISTEN` also showing:

`tcp        0      0 0.0.0.0:3333            0.0.0.0:*               LISTEN `

But with the following error:

```
(gdb) target remote :3333
:3333
: cannot resolve name: Servname not supported for ai_socktype
:3333
: No such file or directory.
(gdb)
```

I found a solution at [https://elinux.org/OpenOCD_Ubuntu_Package](url):

> The package installs config files to /usr/share/openocd/scripts, but [OpenOCD](https://elinux.org/OpenOCD) looks for config files in /usr/local/share/openocd/scripts. Fix that now by creating a symbolic link in /usr/local/share to /usr/share/openocd. In the terminal window, type:

```
cd /usr/local/share
sudo ln -s /usr/share/openocd .
```

Hope that this works for you too!

cr0ck3t, 2023-07-20:
I ran into a similar problem with `netstat -tulpn | grep LISTEN` also showing:

`tcp        0      0 0.0.0.0:3333            0.0.0.0:*               LISTEN `

But with the following error:

```
(gdb) target remote :3333
:3333
: cannot resolve name: Servname not supported for ai_socktype
:3333
: No such file or directory.
(gdb)
```

I found a solution at [https://elinux.org/OpenOCD_Ubuntu_Package](url):

> The package installs config files to /usr/share/openocd/scripts, but [OpenOCD](https://elinux.org/OpenOCD) looks for config files in /usr/local/share/openocd/scripts. Fix that now by creating a symbolic link in /usr/local/share to /usr/share/openocd. In the terminal window, type:

```
cd /usr/local/share
sudo ln -s /usr/share/openocd .
```

Hope that this helps out anyone else who comes across this (or similar) error!

cr0ck3t, 2023-07-20:
Had this happen to me - I was running in VIrtualbox and had not yet added the USB development board to the virtual machine... if you are not using a virtual machine, it may be the the hardware is not mounted.

eldruin, 2023-07-26:
Thank you for finding a fix for this. I would welcome a PR with it for a broader benefit.