rust-embedded/book #269: cargo objdump fails on `-triple` flag when used with a recent LLVM 11

airelil, 2020-10-10:
Running `cargo objdump --bin app --release -- -disassemble -no-show-raw-insn -print-imm-hex` as suggested in [Inspecting](https://rust-embedded.github.io/book/start/qemu.html#inspecting) section of QEMU chapter can produce the following error:
```
cargo objdump --bin app --release -- -disassemble -no-show-raw-insn -print-imm-hex
    Finished release [optimized + debuginfo] target(s) in 0.02s
llvm-objdump: Unknown command line argument '-triple'.  Try: '/home/vk/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--triple'?
llvm-objdump: Unknown command line argument '-disassemble'.  Try: '/home/vk/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--disassemble'?
llvm-objdump: Unknown command line argument '-no-show-raw-insn'.  Try: '/home/vk/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--no-show-raw-insn'?
llvm-objdump: for the --reloc option: may only occur zero or one times!
llvm-objdump: Unknown command line argument '-print-imm-hex'.  Try: '/home/vk/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--print-imm-hex'?
```
It seems that it's fixed with https://github.com/rust-embedded/cargo-binutils/commit/aa5464b4eaacfabe808d00db2e155df3740425a2, but might not been yet published on crates.io (https://github.com/rust-embedded/cargo-binutils/issues/93)
I'm using Ubuntu 20.04 with following llvm-objdump version:
```
bin/llvm-objdump --version
LLVM (http://llvm.org/):
  LLVM version 11.0.0-rust-1.47.0-stable
  Optimized build.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: ivybridge
```
The workaround for me was installing cargo-binutils from GitHub and running it with double hyphens for all flags:
```
cargo install --git https://github.com/rust-embedded/cargo-binutils.git

cargo objdump --bin app --release -- --disassemble --no-show-raw-insn --print-imm-hex
```

adhoore, 2020-10-11:
I have exactly the same problem on windows.

burrbull, 2023-07-28:
Is issue actual?