rust-embedded/book #272: OpenOCD not working on mac

scorpasdzx, 2020-11-07:
I followed all the steps of installing tools required for stm32f3discovery board on Macbook Pro having USB3.0 ports yet when I ran openocd -f interface/stlink-v2.cfg -f target/stm32f3x.cfg I get the following result

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: open failed
in procedure 'init' 
in procedure 'ocd_bouncer'

To my best knowledge, I think its an issue due to usb 3.0 but since I am a newbie I require assistance from sonsummate professionals

jrmoulton, 2020-11-23:
I'm experiencing the exact same issue.

thotsch, 2021-01-02:
I had a similar issue on Big Sur, but with a Nucleo-F439ZI board. In my case, following the hint in #263 helped and I was able to get it to work with the newer stlink config. Maybe try:

`openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg`

Sl1mb0, 2021-06-26:
I am currently experiencing a similar issue, but my error message is a bit different:

Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Error: open failed

There is not any information about why the error occurs either.

KusRoman, 2021-10-29:
> I am currently experiencing a similar issue, but my error message is a bit different:
> 
> Open On-Chip Debugger 0.11.0 Licensed under GNU GPL v2 For bug reports, read http://openocd.org/doc/doxygen/bugs.html WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg Info : auto-selecting first available session transport "hla_swd". To override use 'transport select '. Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD Info : Listening on port 6666 for tcl connections Info : Listening on port 4444 for telnet connections Info : clock speed 1000 kHz Error: open failed
> 
> There is not any information about why the error occurs either.

I'm getting the same sorta issue - unable to find anything.

KoSv, 2023-04-25:
someone got the issue solved?