rust-embedded/book #274: Confusing claim about HashMap in no_std

justinmoon, 2020-12-02:
[This section](https://github.com/rust-embedded/book/blob/master/src/intro/no-std.md#overview) claims the existence of a `collections` crate which can provide `Vec` and `HashMap` implementations.

I don't believe such a crate exists. [Nothing comes up in crates.io](https://crates.io/search?q=collections). `Vec` is provided by `alloc` crate, but `HashMap` is not.

cameronelliott, 2022-11-08:
It is worth noting this page says the reason HM is not available is due to a lack of a secure random number generator:
https://stackoverflow.com/a/65470850/86375

therealprof, 2022-11-09:
That statement is not quite right. There's no `collections` crate but a `collections` module in `alloc` (re-exported in `std`). The fact that `HashMap` (and `HashSet`) are only available in `std` rather than `alloc` due to an implicit requirement of a random generator is annoying but maybe it would be better to simply refer to `hashbrown` for now instead of highlighting an internal detail which is really no help for a user desperately longing for `HashMap`?

cameronelliott, 2022-11-10:
@therealprof did you by chance look at the PR I submitted? Maybe you have some suggestions or improvement ideas for another PR? It actually sounds like a good idea to mention Hashbrown in that section of the docs.

therealprof, 2022-11-10:
I did have a look. I would recommend making two more changes: 
- Change "`collections` crate" into e.g. "`alloc::collections` module"
- Turn the "\** HashMap and HashSet are not available due to a lack of a secure random number generator." remrk into a regular text, recommending to look into `hashbrown` if a `HashMap` or `HashSet` collection type is needed.