rust-embedded/book #278: 2.1 QEMU: cargo objdump gives errors

cocoacrumbs, 2021-01-04:
The embedded rust book in section 2.1 suggests to use this command to disassemble the binary:

```
cargo objdump --bin app --release -- -disassemble -no-show-raw-insn -print-imm-hex
```

However, this results in this error:

```
$ cargo objdump --bin app --release -- -disassemble -no-show-raw-insn -print-imm-hex
    Finished release [optimized + debuginfo] target(s) in 0.01s
llvm-objdump: Unknown command line argument '-disassemble'.  Try: '/home/koen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--disassemble'?
llvm-objdump: Unknown command line argument '-no-show-raw-insn'.  Try: '/home/koen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--no-show-raw-insn'?
llvm-objdump: Unknown command line argument '-print-imm-hex'.  Try: '/home/koen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump --help'
llvm-objdump: Did you mean '--print-imm-hex'?
```

I'm using cargo version 1.49.0
```
$ cargo --version
cargo 1.49.0 (d00d64df9 2020-12-05)
```

The problem seems to be the usage of a single `-`. When I modify the command line to:

```
cargo objdump --bin app --release -- --disassemble --no-show-raw-insn --print-imm-hex
```

works for me.

Thus using `--` instead of a single `-` for the arguments.

I'm not sure if this the intended behavior for cargo or not. Maybe there is a bug in cargo itself parsing the arguments?

joebobfisher, 2021-01-09:
Had the same problem today, with the same fix. Looks like the double hyphens are [added in `master`](https://github.com/rust-embedded/book/blob/master/src/start/qemu.md), but I guess that is not what is published to https://docs.rust-embedded.org/book/start/qemu.html yet.

eldruin, 2021-02-01:
We have a publication issue in our CI:
```
Traceback (most recent call last):
  File "./ghp-import/ghp_import.py", line 10, in <module>
    from dateutil import tz
ImportError: No module named dateutil
```
See [here](https://travis-ci.org/github/rust-embedded/book/builds/756864914#L338)
Follow up issue@ #280

stappersg, 2023-03-11:
> Thus using `--` instead of a single `-` for the arguments.

That is currently published.

[image]

Those who can close this issue, please do.

closed 2023-04-03 by eldruin