rust-embedded/book #279: 2.2 Hardware: References to old example code

cocoacrumbs, 2021-01-04:
The embedded rust book in section 2.2 discusses the following example code:

``` rust
#[entry]
fn main() -> ! {
    hprintln!("Hello, world!").unwrap();

    // exit QEMU
    // NOTE do not run this on hardware; it can corrupt OpenOCD state
    // debug::exit(debug::EXIT_SUCCESS);

    loop {}
}
```

But when we reach the `Debugging` part of the section, we do see this text:

```
Breakpoint 1, main () at examples/hello.rs:15
15          let mut stdout = hio::hstdout().unwrap();
```

This does not correspond to the example code given earlier. I looked a bit in the history of the example code (https://github.com/rust-embedded/cortex-m-quickstart) and the example code was modified in 2018 already to use the `hprintln!` macro (commit `#9c6b290`, "use hprint macros", "japaric committed on Nov 10, 2018").

The second deviation between the book and what I see happening is when setting the break point on `fn main()`:

```
(gdb) break main
```

The text suggests the break point is placed on the first line of the `main` function:

```
Breakpoint 1, main () at examples/hello.rs:15
15          let mut stdout = hio::hstdout().unwrap();
```

This is not the same as I see happening in my setup (`rustc 1.49.0 (e1884a8e3 2020-12-29)`):

```
Breakpoint 1, hello::__cortex_m_rt_main_trampoline () at examples/hello.rs:11
11	#[entry]
```

Thus just before the `fn main()`. When issuing a `next` gdb command (as the book suggests), gdb will simply step over the `fn main()` function and then basically hangs because of the endless `loop ()` inside `fn main()`.

When using the `step` gdb command it's possible to step inside `fn main()' and then execute the `hprintln!` command on its own as intended by the text.

I guess the above deviations might be due more recent `Rust` implementations?

neilpate, 2022-07-09:
I also just ran through this same issue and was similarly confused by this. This was my first exposure to **gdb**, so changing the text to mention being able to `step` would be helpful.

eldruin, 2022-07-11:
Sorry about that. The whole ecosystem has been evolving. We would be glad for PRs that update the instructions.

neilpate, 2022-07-11:
OK, great. I am currently playing around with this in my spare time, so I will try and get a PR done this week.

neilpate, 2022-07-17:
I have created a PR to address this. https://github.com/rust-embedded/book/pull/323

eldruin, 2022-07-25:
Fixed in #323. Thank you!

closed 2022-07-25 by eldruin