rust-embedded/book #283: Section 1.4.4 references unknown files

aThorp96, 2021-02-17:
This section tells you to run `$ openocd -f interface/stlink.cfg -f target/stm32f3x.cfg`, except it doesn't specify where to get those files. Are they part of the project template?

adamgreig, 2021-02-17:
Those files are (or, should be) included with openocd, but if your openocd is a old version (which unfortunately includes the most recent release 0.10.0, from 2017), you might need to try a different filename (`interface/stlink-v2.cfg` perhaps). If you can, I recommend getting the latest nightly version of openocd, there are pre-built images available for various operating systems.

Another option is to use a tool such as PyOCD or [probe-rs](https://probe.rs/), which can also provide a GDB interface, are more up-to-date, and will work with ST-Links as well.

aThorp96, 2021-02-17:
Thanks for your quick response @adamgreig! I may have those files, but I'm not sure. Do you know where `apt` would install the files?

adamgreig, 2021-02-17:
You don't need to provide a path, openocd just knows where to look when you pass it those arguments to the `-f` flag. If you have installed openocd from apt you probably don't have stlink.cfg, though, as the apt version is the latest (super old) release.

In Ubuntu they'd be `/usr/share/openocd/scripts/interface/` and `target/`, you'll probably find `stlink-v2.cfg`. (By the way, you can run `locate stlink-v2.cfg` to rapidly find that sort of thing).

aThorp96, 2021-02-17:
Ah, gotcha. When it didn't find the file I assumed it was a relative path. Thanks! I'll keep the `locate` command in mind :)

aThorp96, 2021-02-17:
I'm not sure whether or not this warrants any changes to the section, if a you need an open ticket for tracking, so I'll let you close it at your discretion. However my initial issue is resolved. Thanks again for the quick help @adamgreig ! Much appreciated

adamgreig, 2021-02-17:
Glad to hear it! I think this is already covered by #277 and #263, so I'll close this one. And, uh, maybe make a note in the book this time...

closed 2021-02-17 by adamgreig