rust-embedded/book #293: 2.2 ~Hardware -- Conflicting info

AlanSwenson, 2021-05-28:
in https://github.com/rust-embedded/book/blob/7349d173fa28a0bb834cf0264a05286620ef0923/src/start/hardware.md

is says : 
```
In examples/hello.rs, make sure the debug::exit() call is commented out or removed. It is used only for running in QEMU.
```

but later on in the page it says :
```
Advancing the program with next should produce the same results as before.

(gdb) next
16          writeln!(stdout, "Hello, world!").unwrap();

(gdb) next
19          debug::exit(debug::EXIT_SUCCESS);
```

Which doesnt make sense to me if you removed it?

I will also add that when removing the debug line you will also need to remove the import of debug as well or it wont load onto the board.  Ill be happy to submit a PR if you let me know which way is correct.

skrapi, 2021-07-09:
I have also ran into this issue. Is there an alternative to using debug::exit(debug::EXIT_SUCCES) when you are running on real hardware?

iFreilicht, 2021-11-02:
It also seems like that output is outdated in general. The line numbers don't match, and it should say `hprintln`, not `writeln`.