rust-embedded/book #300: OpenOCD not connecting to target

MartBent, 2021-09-01:
Hello,

I followed the windows installation instructions from the embedded rust book: https://docs.rust-embedded.org/book/intro/install/verify.html

Now when I try to run the .cfg files with OpenOCD I get this output: (I am using the STM32WL55JC1chip on the nucleo development board)

PS H:\> openocd -f interface/stlink.cfg -f target/stm32wlx.cfg
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 500 kHz
Info : STLINK V3J7M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.266045
Error: init mode failed (unable to connect to the target)

When holding down the reset button the hardware breakpoints are detected:

PS H:\> openocd -f interface/stlink.cfg -f target/stm32wlx.cfg
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 500 kHz
Info : STLINK V3J7M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.267643
Info : stm32wlx.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32wlx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: jtag status contains invalid mode value - communication failure
Polling target stm32wlx.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect

But then it ends up crashing when I let go of the reset button.

What is the cause of this issue? I have already tried running it as administration and that didn't seem to solve anything.

Thanks and regards,

Mart

adamgreig, 2021-09-06:
You might need to try connecting under reset, in case the current firmware is doing something that disables debug access (e.g. sleep mode). With openocd, you could make a new .cfg file (and run it with -f) containing something like:

```
reset_config srst_only srst_nogate connect_assert_srst
```

but, you might also check out https://probe.rs, as tools like `cargo flash` can simply be given `--connect-under-reset. For example, after `cargo install cargo-flash`, `cargo flash --release --chip stm32wle5jcix --connect-under-reset` from your firmware project. It looks like probe-rs doesn't yet have the WL55 series but perhaps the WLE5 are close enough to get started.

MartBent, 2021-09-08:
> You might need to try connecting under reset, in case the current firmware is doing something that disables debug access (e.g. sleep mode). With openocd, you could make a new .cfg file (and run it [...]

Altering the reset in the cfg file fixed the issue. Thank you

closed 2021-09-08 by MartBent