rust-embedded/book #302: Installation of cargo-generate on Ubuntu fails owing to missing OS packages

rod-chapman, 2021-10-01:
I am trying to follow the installation instructions in section 1.4 on a stock Ubuntu Beaver installation.

The "cargo install cargo-generate" command fails owing to missing OS packages "libssl-dev" and "pkg-config".

I had to do "sudo apt install libssl-dev pkg-config" before it would work, so this should probably be mentioned in section 1.4 somewhere.

eldruin, 2021-10-06:
Good catch. Could you open a PR adding this information?

rod-chapman, 2021-10-06:
I'm not sure... Section 1.4 is supposed to be OS independent, with Linux-specific stuff following in section 1.4.1, so perhaps these sections need to be re-ordered a bit? I'm not confident enough to make such a change. Who has editorial/review control over this section?

rod-chapman, 2021-10-06:
Do I need to fork this repo first, then propose a PR from there?

rod-chapman, 2021-10-06:
OK.. I will give it a go...

PR #304 (2021-10-06): Update section 1.4 to mention that some additional packages required
...before cargo-generate can be installed.

Fixes #302
closed 2021-10-06 by bors[bot]