rust-embedded/book #306: Small sections about using math library

robamu, 2021-11-08:
Correct me if I am missing any information, but I was not able to use some standard float operations like `floor` and `ceil` and I required those when calculating register values for a HAL. I now added the `libm` dependency and was able to use the `floorf` functions for my purposes, but I think not being able to use functions like `floor` is related to having the `#[no_std]`  attribute on the application?

If that is the case, it might be worth an idea to add a small section related to how to load more advances math libraries.

eldruin, 2021-11-10:
That is correct. It would be great if you could contribute that small section about this in a PR.

robamu, 2021-11-10:
Unfortunately, I'm not an expert (yet), I only stumbled upon this when trying to do floating point calculation on a MCU. I could try to get some information about which math operations are allowed and which ones are reserved to `std` . I might do some research. Maybe someone already compiled a list or something similar? Then I could try to contribute a small section about this topic.

robamu, 2021-11-11:
A related RFC can be found here: https://github.com/rust-lang/rfcs/issues/2505

PR #308 (2021-11-14): added math chapter
See #306 . I tried to gather some more information about available crates if more complex functionality is required and added links to them as well.

Fixes #306
closed 2021-11-14 by bors[bot]