rust-embedded/book #325: Apple Silicon can't install gdb, lldb instructions needed?

aguynamedben, 2022-07-30:
Hi, this book is great, but I have an Apple Silicon (M1) computer and `brew gdb` simply fails with:
```
$ brew install gdb
gdb: The x86_64 architecture is required for this software.
Error: gdb: An unsatisfied requirement failed this build
```

 It seems that gdb is just not supported on Apple Silicon computers.

I'm new to gdb, lldb, and debuggers. It would be cool if the QEMU chapter could be updated to show how Apple Silicon users can use lldb to debug embedded Rust apps. Thank you!

aguynamedben, 2022-07-30:
I'm not sure how it works under the hood, but QEMU and hardware debugging works if you use VS Code + cortex-m-quickstart + these instructions:

https://christopherjmcclellan.wordpress.com/2019/12/31/debugging-rust-cortex-m-with-vs-code-take-2/

toblux, 2022-08-28:
Using `arm-none-eabi-gdb` instead of `gdb` works for me. It's installed with `arm-none-eabi-gcc`. See https://docs.rust-embedded.org/book/intro/install/macos.html

Velnbur, 2024-07-13:
`lldb` worked for me [thanks by this stack overflow comment](https://stackoverflow.com/a/69784630)