rust-embedded/book #326: I think this repository can support more languages

Jzow, 2022-08-09:
have this plan at present?

Jzow, 2022-08-09:
#126

eldruin, 2022-08-09:
We would be glad for translations of the book into Chinese and any other language!
[mdbook does not have multilingual support yet](https://github.com/rust-lang/mdBook/issues/5) so I think an easy way to do this for now would be to fork this repository and then translate it. You can then publish it to GitHub pages and we will link it here.
This is what @tomoyuki-nakabayashi has done for the [Japanese version of the discovery book](https://github.com/tomoyuki-nakabayashi/discovery)
Does that sound reasonable to you?
We can help you if some steps do not work.

mgeisler, 2023-05-12:
Hi there, I wrote mdbook-i18n-helpers and we've been using it for some months now for [Comprehensive Rust 🦀](https://google.github.io/comprehensive-rust/running-the-course/translations.html). I would be happy to help with setting it up for this book.

I see that there are already two hand-written translations available. Getting those converted back to the Gettext PO format will be a non-perfect process. I can try writing a convert tool which takes two Markdown files as input and outputs a PO file. If the translations have kept the Markdown structure intact (same number of paragraphs, same number of list items, ...), it should be possible to convert them back to a PO file.

eldruin, 2023-05-12:
Sounds great @mgeisler, thank you. It would be great to draw more people into embedded Rust :)
However, please beware that we generally do not have much time to invest, so the setup would need to be easy to manage.

mgeisler, 2023-05-12:
> Sounds great @mgeisler, thank you. It would be great to draw more people into embedded Rust :)

Yeah, I would hope so!

> However, please beware that we generally do not have much time to invest, so the setup would need to be easy to manage.

Makes sense! In general, the setup I use for the course favors long-term maintainability for some simplicity. The easiest no-setup-needed idea is to clone the repository and do the translations directly inside the English files. However, it's hard to impossible to track changes to the original files that way.

The Gettext approach helps here since it decouples the translation from the source. You extract source text into PO files and translators work on those. When you publish a translation, you read the translation from the PO files. The PO files are small version-controlled databases and there is tooling available which can merge new strings into these databases. This is described in detail in [mdbook-i18n-helpers](https://github.com/google/mdbook-i18n-helpers/).

mgeisler, 2023-05-16:
> @mgeisler Could you write CI for this repository for `pot` file update and release translated books?

Yes, I would be happy to do this! It might be a week or two before I get to it.

mgeisler, 2023-05-16:
> It might be a week or two before I get to it.

I nerd-sniped myself and couldn't help but working on it... please see #356 :smile:

Jzow, 2023-05-17:
Excuse me, I have two questions. 

Firstly, I have roughly reviewed the description of @mgeisler The [mdbook-i18n-helpers](https://crates.io/crates/mdbook-i18n-helpers) warehouse is designed to help md convert to po, and I don't quite understand what the po here means. What is its purpose?

 The second question is, do we need to wait for this plugin to complete before translating, and do we need a manual before translating.

burrbull, 2023-05-17:
`.po` is translation file for https://ru.wikipedia.org/wiki/Gettext

Supported by several editors. For example
https://poedit.net/

Jzow, 2023-05-17:
> `.po`[是https://en.wikipedia.org/wiki/Gettext](https://en.wikipedia.org/wiki/Gettext)的翻译文件[](https://en.wikipedia.org/wiki/Gettext)
> 
> 得到多位编辑的支持。例如 https://poedit.net/

thanks!

mgeisler, 2023-05-27:
Hey all, I just wanted to say that offer in #356 still stands:

> [...] I hope people can give it a try. If someone (such as @DownyBehind from #350 and @Jzow from #326) wants to play with a Gettext-based translation, then they're free to open PRs against https://github.com/mgeisler/rust-embedded-book. That repository can be a temporary "translation playground" so that you don't have to deal with anything in this repository.

I have no connection with this repository and I think it's up to the Embedded Rust community to decide if they want to embark on a translation. If you do want to try, then feel free to add stuff https://github.com/mgeisler/rust-embedded-book — I'm happy to give people write access so they can experiment freely there. The results will show up in the #356 PR automatically :slightly_smiling_face: 

For Comprehensive Rust, we've just embarked on adding Chinese (Simplified) and Chinese (Traditional) translations and the [PRs are coming in now](https://github.com/google/comprehensive-rust/pulls?q=is%3Apr+is%3Aopen+zh-). That might help show the system in action.

eldruin, 2023-05-30:
Thank you @mgeisler! We will discuss this again in tonight's meeting.

eldruin, 2023-05-31:
There is some support to go through this route, however, the [discussion](https://github.com/rust-embedded/wg/blob/master/minutes/2023-05-30.md#agenda) [yesterday](https://libera.irclog.whitequark.org/rust-embedded/2023-05-30#34370143) highlighted that there are a couple of missing pieces in this PR:
1. Extraction of the `po` templates
2. Merging of the `po` files

Furthermore, the upstream Rust docs also link to externally hosted copies of the book so if we used `po` files, we would probably have to coordinate with the upstream Rust docs.

burrbull, 2023-06-07:
> renderer on-demand

This is the issue.
If you don't see absent translations you don't know you need to actualize it.
I would better see incomplete translation then outdated one.

mgeisler, 2023-06-07:
> > renderer on-demand
> 
> This is the issue. If you don't see absent translations you don't know you need to actualize it. I would better see incomplete translation then outdated one.

Okay, I think I understand what you mean!

The `messages.pot` file is not used when rendering a translation into HTML, but it is used when updating the `xx.po` file for the xx language.

Where you hoping to have all the PO files updated in lockstep when the English source changes? Something like a GitHub action which runs the following?

```sh
MDBOOK_OUTPUT='{"xgettext": {"pot-file": "messages.pot"}}' mdbook build -d po
for pofile in po/*.po; do
    msgmerge --update $pofile po/messages.pot
done
git add po/
git commit -m "Sync all PO files"
```

That would be possible, but awkward: updating the PO files on behalf of the translators takes away the possibility for the translators to see the diff after a `msgmerge`.

What I would suggest instead would be to run

```sh
for pofile in po/*.po; do
    msgmerge --update $pofile po/messages.pot --quiet
    echo -n "$f "
    msgfmt -o /dev/null --stat $pofile
done
```

in a GitHub action after every merge. The output will show the number of translated and untranslated messages for each language.

That's a passive way to have the information available. If you want a way to notify translators, then you have to look into using a translation platform such as https://www.transifex.com/. I used them years ago and it looks like they still support Gettext PO files.

Jzow, 2023-08-01:
I'm glad to see this translation conversion tool, but currently it may require evaluation from members of the embedded resource group. Before officially using it, the solution I can think of is to submit an **issue** or **pr** to other translation repositories after synchronizing updates, to alert maintainers of other language repositories for updates

Jzow, 2023-08-01:
> I would better see incomplete translation then outdated one.

Agree, this is the majority of reading habits

mgeisler, 2023-08-02:
>> > renderer on-demand
>> 
>> This is the issue. If you don't see absent translations you don't know you need to actualize it. I would better see incomplete translation then outdated one.
>>
> Agree, this is the majority of reading habits

Reading this again, I realize that I might not have explained myself well above. If you look at https://google.github.io/comprehensive-rust/ko/, you will see that the page has a mix of Korean and English. This is because we re-publish all languages whenever something changes in `main`. So if a typo is fixed in the English source text, we immediately invalidate the affected paragraph in all translations.

So readers of the book certainly notice that something has become outdated :slightly_smiling_face: 

I'm considering changing this approach since it's hard work for translators to keep up to date. One approach would be to reuse an existing translation as long as it is more than NN% complete. If set to 95%, the translation would be allowed to degrade from 100% down to 95% after which it would stop updating. Degrading means that more and more paragraphs revert back to English. Stop updating means that the text is frozen at this point — so typo fixes won't flip paragraphs back to English, but actual fixes and additions won't show up either.

I don't know if this was clear in the above?

> Before officially using it, the solution I can think of is to submit an **issue** or **pr** to other translation repositories after synchronizing updates, to alert maintainers of other language repositories for updates

I think this is a good idea. The whole thing is very independent of the source repository — interested contributors can easily fork the repository and maintain their translations using mdbook-i18n-helpers without any help needed from this repository. If they like the system, then it's easy to merge things back later.