rust-embedded/book #327: How would one cite this book in an academic resource

cvvletter, 2022-08-26:
See title, preferably in biblatex style.

cvvletter, 2022-08-26:
This is what I came up with:
```
@book{embedded-rust,
  title = {The Embedded Rust Book},
  author = {},
  date = {},
  url = {https://docs.rust-embedded.org/book/},
  urldate = {2022-08-26},
}
```
Maybe the author would be 'Rust on Embedded Devices Working Group', but what could I put as date or the likes?

adamgreig, 2022-08-26:
That seems OK. I probably would just leave a date accessed, the same as you might for a URL, or the date of the most recent commit when you viewed it. For author, "Rust Embedded Working Group & Contributors" or something like that would be fine.

cvvletter, 2022-08-29:
Thank you. I changed it to the following:
```
@book{embedded-rust,
  title = {The Embedded Rust Book},
  author = {{Rust on Embedded Devices Working Group} and others},
  date = {}, % intentionally left empty
  url = {https://docs.rust-embedded.org/book/},
  urldate = {2022-08-26},
}
```
which is correctly handled (in IEEE format) as: Rust on Embedded Devices Working Group _et al._, _The Embedded Rust Book._ [Online]. Available: `https://docs.rust-embedded.org/book` (visited on 26/08/2022).

Would it be useful if I add the preferred citation at the end of the Introduction in a pull request?

adamgreig, 2022-08-29:
Looks good. Maybe the best way to add it to the repository would be via GitHub's built-in support for citations:

https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files

Ideally you'd write a CFF file with the same info, and then GitHub can parse it and render it to a bibtex or other formats depending on what people want. It looks like you could also just stick a `CITATION.bib` file in the repo root and it will at least offer 'Cite this repository' as an option, though without parsing or converting the file.

cvvletter, 2022-09-12:
I chose to only PR the .bib file, due to the .cff file not supporting "_et al._" nor empty date fields. Feel free to let me know your thoughts.

commit 4ce51cb7 (2022-09-15): Merge #329

329: Create CITATION.bib (as per #327) r=eldruin a=cvvletter

which allows for a pre-made Github citation button.

Fixes #327
PR #329 (2022-09-15): Create CITATION.bib (as per #327)
which allows for a pre-made Github citation button.

Fixes #327
closed 2022-09-15 by bors[bot]