rust-embedded/book #350: Is there a procedure required to translate the repository into another language?

DownyBehind, 2023-05-04:
I already read #326.

I wonder if it is okay to proceed with the procedure left by @eldruin  on the issue. #326 

I want to translate it into Korean.

thank you.

eldruin, 2023-05-04:
Yes, that is still the right procedure. We might do something like having `po` files at some point in the future since that approach has now appeared but that requires some time to get right and I for one would not have the time to do it.

DownyBehind, 2023-05-05:
Thank you for your reply.
Then, as I said, I will add a link after translating it into Korean.

almightychang, 2023-07-11:
How is it going though? Are you still on translation??

eldruin, 2023-07-12:
@romancardenas and I are [translating this book into Spanish through transifex](https://explore.transifex.com/rust-embedded/rust-embedded-book/), which works pretty nicely.
Let me know if you want me to start a translation into Korean or any other language.

DownyBehind, 2023-07-12:
@almightychang I stopped translating for a while because of my other work, but I will do it again soon. But there seems to have been a lot of talk. By the way, are you interested in Korean translation?

@eldruin Do I have to translate in a new way instead of the way you mentioned before (making the translation into github page and adding Link to master branch)?

eldruin, 2023-07-12:
We are strongly considering taking the approach in #356 and thus we started with a test with a Spanish translation.

DownyBehind, 2023-07-12:
@eldruin Then can I proceed with the Korean translation in that way? Or should I wait until you guys finish the test?

eldruin, 2023-07-31:
Sorry @DownyBehind, I wanted to discuss this in our weekly meeting again but have been unable to so far. I will try again tomorrow.
I do not want to waste your time and effort so please hold on for a bit until we try the new approach out.
My status is that we have a roughly 50% Spanish translation [here](https://app.transifex.com/rust-embedded/rust-embedded-book) (about 10% of the snippets are not translatable / not worth translation like console outputs). So far the translation experience has been pretty smooth.

DownyBehind, 2023-08-01:
@eldruin  Thank you for your answer. Then I'll be waiting for the results of your new attempts

mgeisler, 2023-08-14:
> My status is that we have a roughly 50% Spanish translation [here](https://app.transifex.com/rust-embedded/rust-embedded-book) (about 10% of the snippets are not translatable / not worth translation like console outputs). So far the translation experience has been pretty smooth.

Hi @eldruin, just for my curiosity: are you using Transifex to translate a PO file generated with mdbook-i18n-helpers? I used Transifex ~10 years ago for this — but perhaps the platform now also supports translating Markdown files directly?

If you do use mdbook-i18n-helpers, you should be aware of https://github.com/google/mdbook-i18n-helpers/pull/46. The `main` branch already has a much better text extraction system than what is released in version 0.1.0. The PR linked has a normalization tool which can will let you convert existing PO files (generated by version 0.1.0) to the new format. The new format removes the Markdown syntax. As an example, `# Foo` becomes `"Foo"` in the PO file and the text for tables is extracted on a cell-by-cell basis.

I hope this is useful!

mgeisler, 2023-08-23:
I've released version 0.2 of mdbook-i18n-helpers — this includes the much better text extraction which will unwrap paragraphs and remove unnecessary Markdown syntax in the translation files. The normalization tool is also included so that you can keep existing translations intact (if you have any, we have ~18 for Comprehensive Rust). Please let me know of any problems you might run into! Thanks :slightly_smiling_face:

leoliu0605, 2025-07-28:
Hi @mgeisler @eldruin, is there any progress regarding the process of contributing translations? I hope to submit a Traditional Chinese translation.