rust-embedded/book #353: Links to other resources

mgeisler, 2023-05-08:
Hi there,

I maintain [Comprehensive Rust 🦀](https://google.github.io/comprehensive-rust/) and we've recently added a full day about [Bare-metal Rust](https://google.github.io/comprehensive-rust/bare-metal.html).

Would there be interest in having a section in your book with links to other resources, such as this course?

I created [such a page](https://google.github.io/comprehensive-rust/other-resources.html) for my course and I've found it very helpful since it gives me something to point students to when they want to learn more.

eldruin, 2023-05-10:
Hi, that would definitely be welcome here.
Would you mind opening a PR?

mgeisler, 2023-05-12:
Thanks @eldruin, I created #354 to add a link!

commit dc85da5c (2023-05-12): Add link to Comprehensive Rust 🦀

Fixes #353.
commit f2aed2fe (2023-05-12): Merge #354

354: Add link to Comprehensive Rust 🦀 r=eldruin a=mgeisler

Fixes #353.
PR #354 (2023-05-12): Add link to Comprehensive Rust 🦀
Fixes #353.
closed 2023-05-12 by bors[bot]