rust-embedded/book #355: Tooling page is significantly outdated

burrbull, 2023-05-12:
https://github.com/rust-embedded/book/blob/master/src/intro/tooling.md

Nothing about `probe-rs` project; old Rust version, etc.

eldruin, 2023-05-15:
Indeed. Thanks for noticing.
To anybody reading this, please feel free to send a PR updating it.
[Here](https://docs.rust-embedded.org/discovery/microbit/03-setup/index.html) is an example of a more up-to-date setup

jonathanpallant, 2023-09-20:
Yeah I came across this today and went to write up basically the same ticket.

jonathanpallant, 2023-09-20:
I think the steps are now:

* Install `rustup`
* Add appropriate targets (i.e. pre-compiled rust libraries) for your MCU
* `cargo install probe-run --features=cli`

No OpenOCD required, nor arm-none-eabi-gcc.

You could still list GDB (or `arm-none-eabi-gdb`) as an optional thing, but you should probably call it through the `rust-gdb` wrapper script (first setting `RUST_GDB=arm-none-eabi-gdb` to it knows what to call).

Barbacamanitu, 2023-09-30:
> I think the steps are now:
>
> * Install `rustup`
> * Add appropriate targets (i.e. pre-compiled rust libraries) for your MCU
> * `cargo install probe-run --features=cli`
>
> No OpenOCD required, nor [...]

Is there a guide somewhere on how to set this up? I just got my first STM chip to follow the embedded Rust book, and I'm not sure where to start or how to follow along since the tooling is outdated now. For now I guess I'm going to follow the current version of the book exactly just so I can get some code flashed to my board, but it would be nice to use the newer tools.

probe-rs sounds neat, but it also seems like its a library? I'm not sure how I'd use it to flash and debug. Any help would be appreciated.