rust-embedded/book #375: document futures size explosion pitfall and mitigations

spookyvision, 2024-08-23:
As explained e.g. [here](https://swatinem.de/blog/future-size/), futures can easily turn into unpleasant memory hog surprises.
One possible easy way to investigate problematic code is to add a `future-size-threshold = 100` (or whichever threshold seems reasonable) clippy lint.

eldruin, 2024-09-19:
We would welcome a PR explaining this. Please feel free to write it if you have the time.

spookyvision, 2024-10-07:
I don't feel qualified enough to do a detailed writeup, so would rather see somebody else spearheading this. If nobody else takes it up, I can try and come up with 1-2 paragraphs…

spookyvision, 2025-11-23:
more from that discussion, quoting & paraphrasing `@fg-cfh`:
- useful to figure out the biggest offenders: `cargo +nightly rustc --release -p ... --target ... --bin ... -- -Zprint-type-sizes`
- potential mitigations hand-optimize futures as much as possible
  -  remove arguments
  - make state static
  -  turn async blocks into futures with combinators.

spookyvision, 2025-11-27:
I have caught up with the discussion in Matrix. I think what @adamgreig suggested makes sense: provide a brief note for now, and expand on that later. My suggestion would be to have a separate issue for the "brief for now" approach, because the current issue contains a collection of information relevant for a longer writeup

BartMassey, 2025-11-27:
I guess I don't quite know what to put in the brief note. "On rare occasions you may find that your embedded program runs out of memory because of a misfeature of the current implementation of Rust async/await that can cause ridiculously large structures to be created as the result of [???]. You can quickly recognize this situation by [???], and deal with it by [???]." If I could fill in the missing pieces here, it would make sense to put that in. As it is…

fg-cfh, 2025-11-29:
> I'm not sure why I've never heard of it.

You're not the only one. A warning in the embedded book would have saved me a lot of work.

> I guess I don't quite know what to put in the brief note.

I suggest that we mention the tooling that allows developers to observe the impact of using async/await. That's what would have helped me to take better decisions early on.

Notably:
- cargo size
- cargo bloat
- -Zprint-type-sizes together with top-type-sizes.

I also suggest that this writeup merits a link from the embedded book:

https://rust-lang.github.io/wg-async/vision/submitted_stories/status_quo/barbara_carefully_dismisses_embedded_future.html

The blog post you cited should also be linked IMHO, at least as long as those bugs aren't fixed.

IMO, it would also be appropriate to mention, that Aaron Turon's original ["futures are zero-cost"](https://aturon.github.io/blog/2016/08/11/futures/) related to hand-written futures and futures-rs combinators, not to async/await blocks.

Furthermore I suggest we link to the following issue search so everyone can read up on known problems before taking an architecture decision:

https://github.com/rust-lang/rust/issues?q=state%3Aopen%20label%3A%22A-async-await%22%20label%3AI-heavy

> Apparently most of the time the compiler fixes this by inlining `.await` and optimizing the result.

I don't think this is true, at least I cannot confirm from real-world experiments.

Desugaring an async call hierarchy results in nested enums (state machine state) + nested function calls (mostly simple nested match-blocks on those enums).

LLVM is able to inline nested match blocks but not nested enums. IIUC, from LLVM's viewpoint those enums are just arbitrary data structures with a fixed memory layout that LLVM is not entitled to tamper with. What blows up stack and flash size is the data structures, not the glue code.

For those interested in more detail: Insight can easily be gained from -Zprint-type-sizes on any application that uses async/await blocks/fns:

- Even in --release mode, each async nesting level inevitably seems to add an additional enum nesting level, capturing a copy of each local upvar at MIR level (no matter whether that upvar is later inlined by LLVM, required across the first await point, already captured further up in the call stack or even used at all, including [&[mut]]self).

- Structures (including self passed by value) seem to be captured wholly, not at field level or as a reference. Typical optimizations like instantiating a result struct one level up don't seem to apply to async code AFAICS.

- Upvars actually used in the function body will then be captured _again_ according to the usual closure capturing rules.
 
- Exponential blowup stems from the fact that determinant + upvars + captured vars from the next lower level are recursively added as a sub-field to the enum one level up, even if only a single .await point with none or only trivial pre-/post-processing around it exists one level down.

- In a library context (as opposed to application context), hand-written futures do _not_ work in practice (at least not for us). You basically loose all the benefits of `impl Future` and end up with as many generic arguments as you have nesting levels.

You can work around some (but not all) of that bloat but IMHO only at the price of severly reduced ergonomics and language feature coverage. One commentor went so far as to recommend not to nest async functions/blocks at all (and even that will not be a zero-cost abstraction).

IIRC, the officially recommended tokio workaround is to systematically Box any large future along the call stack which breaks the blow up. That's obviously not a good solution for a no-alloc environment (and maybe not even on alloc with all the additional allocations and moving around of memory at runtime. I haven't measured that approach on a real system, though, because it is off-limit for our use case anyway).

Update: I want to say this clearly - Most of this hurts applications much less than libraries. As the author of a highly space+time sensitive library, I want to be zero-cost as much as possible. As an application author, I'm satisfied with "good enough". So this doesn't invalidate most async embedded use cases of course.